miaoermua/CatWrt #7: 有线/无线驱动解决计划

miaoermua, 2023-02-03:
为响应大家对 CatWrt 更多应用范围，有需增加驱动的小伙伴，留言一下所需的有线无线的机型和芯片，针对 x86 和 ARM 版本的 CatWrt 扩展建议，wireless 系列请勿参与。

如格式: 无线 - 绿联 AC650M - RTL 8811CU

计划优先 x86 平台更新

目前已知 ARM 平台友善 R2s lan 口无法连接，wan 口可以，已经在跟进处理；英特尔系列 AP 模式目前已经测试成功(只能开启 2.4G)

jiangxin0867, 2023-02-03:
无线 - 腾达 AC650M - RTL 8811CU

AEnjoy, 2023-02-03:
Those packages that are Intel AX210 needed:
kmod-iwlwifi wireless-regdb wireless-tools **iwlwifi-firmware-ax210** wpa-supplicant hostapd;

And the other Intel wireless firmware packages could be found at Luci-System-Packages-Research, typing iwlwifi.

miaoermua, 2023-02-03:
> 

还有 kmod-cfg80211 和 kmod-mac80211

miaoermua, 2023-02-03:
已有测试版体验 https://github.com/miaoermua/CatWrt/releases/tag/v23.2
还差无线驱动部分，将在最近完成并且推送公测版

miaoermua, 2023-02-15:
目前已经完善部分驱动 https://github.com/miaoermua/CatWrt/commit/ae3bae4e8ea02c3d0e23c98d13a3b54c9ac17fc8

LCW0NJUPT, 2023-04-02:
无线 - orangePi R1Plus - RTL 8821CU

Rikkaawa, 2023-05-16:
无线 - 华硕ChromeBox CN62 - Intel AC 7260

closed 2023-08-08 by miaoermua