miaoermua/CatWrt #15: [ Feature ] 考虑为红米AX6000固件增加SFTP功能

Kia678, 2023-04-14:
add vsftpd openssh-sftp-server

miaoermua, 2023-04-14:
可以有，看后续

miaoermua, 2023-04-15:
https://github.com/miaoermua/CatWrt/search?p=3&q=sftp

其实是有的，早上赶事

miaoermua, 2023-04-15:
[](url)

closed 2023-04-15 by miaoermua

Kia678, 2023-04-17:
> https://github.com/miaoermua/CatWrt/search?p=3&q=sftp
> 
> 其实是有的，早上赶事

[image]

提示这个错误，你看一下

Kia678, 2023-04-23:
麻烦看一下上面这个错误

miaoermua, 2023-05-15:
> 麻烦看一下上面这个错误

更新列表了吗，在终端输入安装命令输出的内容发上来。

miaoermua, 2023-05-18:
> > > 麻烦看一下上面这个错误
> > 
> > 
> > 更新列表了吗，在终端输入安装命令输出的内容发上来。
> 
> [image]
> 
> 你看看，要怎么解决SFTP的问题，红米AX6000固件下个版本能否集成SFTP功能

你压根就没用我的源。。。

Kia678, 2023-05-18:
> > > > > > 麻烦看一下上面这个错误
> > > > > 
> > > > > 
> > > > > 更新列表了吗，在终端输入安装命令输出的内容发上来。
> > > > 
> > > > 
> > > > [image]
> > > > 你看看，要怎么解决SFTP的问题，红米AX6000固件下个版本能否集成SFTP功能
> > > 
> > > 
> > > 你压根就没用我的源。。。
> > 
> > 
> > 你的源在哪里看？怎么替换
> 
> https://www.miaoer.xyz/posts/network/catwrt-expand

谢谢，openssh-sftp-server 已经安装好了。红米AX6000固件下个版本能否内置openssh-sftp-serve进去？

miaoermua, 2023-05-18:
不能要瘦身