miaoermua/CatWrt #18: IPv6的问题

Kia678, 2023-04-17:
### 问题: 
目前WAN6接口可以获取到光猫下发的ipv6，也能在catwrt的ssh里面ping通ipv6
但是LAN口下的设备无法使用ipv6(包括无线WiFi和有线)

### 能否复现: 
能

### 版本信息

设备:Redmi AX6000

版本号: Catwrt23.2

miaoermua, 2023-04-17:
看看 lan 的 ipv6，在 catwrt 中 ping 一下外部 ipv6 域名或者地址 ping `6.ipw.cn` 看看返回是否是 ipv6 地址。
查看是否有 ipv6–pd；
另外的防火墙问题仅影响外部对你的访问和无限转圈圈后续修复。

Kia678, 2023-04-18:
能在catwrt中ping通外部ipv6
仅仅只是LAN下的设备无法ping通ipv6
没有ipv6-pd

miaoermua, 2023-04-18:
> 能在catwrt中ping通外部ipv6
> 但是LAN下的设备无法ping通外部ipv6
> 没有ipv6-pd

如果你是光猫桥接的就不需要 ipv6–pd，如果你是光猫拨号则需要，平替就是 nat6。

如果光猫拨号你的设备需要你的运营商或者校园网支持 ipv6–pd 呢。

如果 catwrt 拨号可以查看一下无线 ap 是否开启了 ap 模式(有线桥接)，以及设备设备是否因为游戏加速器优化掉了 ipv6 选项。

Kia678, 2023-04-18:
我上面已经说了是光猫拨号
另外，在catwrt里面，可以ping通外面ipv6域名或网址，也可以正常访问ipv6域名
但是catwrt的LAN下面的设备无法ping通外部ipv6
光猫的ipv6本身没有任何问题，光猫的LAN网线直接插电脑网口是可以用ipv6的

miaoermua, 2023-04-18:
> 我上面已经说了是光猫拨号
> 另外，在catwrt里面，可以ping通外面ipv6域名或网址，也可以正常访问ipv6域名
> 但是catwrt的LAN下面的设备无法ping通外部ipv6
> 光猫的ipv6本身没有任何问题，光猫的LAN网线直接插电脑网口是可以用ipv6的
> 给个Telegram联系方式吧

tg 在博客有，如果没有 ipv6–pd 是不行的，可以考虑桥接

Redleaves5, 2023-05-15:
可以试试我的方法https://www.bilibili.com/read/cv14169428
并不对所有OpenWrt固件有效,但值得一试