miaoermua/CatWrt #20: 新路由3 CatWrt v22.12缺少USB和EXT4/EXFAT支持

Redleaves5, 2023-05-15:
### 问题: 
新路由3 CatWrt v22.12缺少USB和EXT4/EXFAT支持

### 能否复现: 
能

### 版本信息
CatWrt v22.12
设备:
新路由3

[image]

- [ ] 我已按照规定格式编写，没有的内容或备注请删除
- [x] 我使用的是 CatWrt

miaoermua, 2023-05-15:
需要自己添加，具体看 [博客](https://www.miaoer.xyz/posts/network/catwrt-expand)，默认不带因为会影响 2.4G 无线

Redleaves5, 2023-05-15:
> 需要自己添加，具体看 [博客](https://www.miaoer.xyz/posts/network/catwrt-expand)，默认不带因为会影响 2.4G 无线

opkg install 安装了以下包：
[image]
但是没有效果

miaoermua, 2023-05-15:
> > 需要自己添加，具体看 [博客](https://www.miaoer.xyz/posts/network/catwrt-expand)，默认不带因为会影响 2.4G 无线
> 
> opkg install 安装了以下包：
> [image]
> 但是没有效果

需要先进行 opkg update，需要完整命令输出结果才能诊断

miaoermua, 2024-08-24:
在新版添加了 kmod-fs-ext4，除了 exfat，其他都齐了

miaoermua, 2024-08-26:
[image]

已支持，所以关闭了，如有需要随时打开 issues 提问

closed 2024-08-26 by miaoermua