miaoermua/CatWrt #29: 如何进行DSA VLAN的设置

iseku, 2023-09-12:
### 问题: 
目前发布的版本未包含完整的DSA网络配置功能, Luci菜单 network 下没有 devices 的配置项, 无法进行vlan配置, 使用自定义接口配置的软件VLAN无法处理tag和untag, 请问在后续版本中是否会支持相关内容? 或者能单独发布一个手工配置说明.

### 能否复现: 
能

### 版本信息

设备:
Xiaomi Redmi Router AX6000
版本号: 
CatWrt.v23.8.mt798x / OpenWrt R23.7.7 / LuCI Master (git-23.198.59743-991daf5) 

*终端执行 `/usr/bin/catwrt-update` 或 `cat /etc/banner` 再或者提供刷机文件名*
### 日志: 

- [ ] 我已按照规定格式编写，没有的内容或备注请删除
- [ ] 我使用的是 CatWrt

*禁止 issues 上传别人固件的错误信息，日志长的可以分段发送，请确认在提出问题前你已经阅读过博客*

iseku, 2023-09-15:
通过手工配置network启用了bridge vlan filtering, 可以正常使用vlan了, 但是luci的网络配置出现不支持的网络协议, 看了下opkg源设置, 应该是集成的老版本luci-base, 不支持DSA配置, 希望能在下个版本解决这个问题.
network配置:
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'xxxx:xxxx:xxxx::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'

config device
	option name 'eth0'
	option macaddr 'xx:xx:xx:xx:xx:xx'

config interface 'lan'
	option device 'br-lan.1'
	option proto 'static'
	option netmask '255.255.255.0'
	option delegate '0'
	list dns '172.16.71.254'
	option ipaddr '172.16.71.253'
	option gateway '172.16.71.254'

config interface 'wan'
	option proto 'dhcp'
	option device 'br-lan.2'

config bridge-vlan
	option device 'br-lan'
	option vlan '1'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan:u*'

config bridge-vlan
	option device 'br-lan'
	option vlan '2'

config bridge-vlan
	option device 'br-lan'
	option vlan '3'
	list ports 'lan2'
	list ports 'wan:t'

config interface 'iptv'
	option proto 'dhcp'
	option device 'br-lan.3'
	option metric '20'

config device
	option name 'eth1'
	option macaddr 'xx:xx:xx:xx:xx:xx'

问题截图:
[image]

miaoermua, 2023-10-25:
那应该就是 luci 版本问题了，没玩过

closed 2023-11-13 by miaoermua