miaoermua/CatWrt #32: 请针对mt798x的后续更新中，添加对wireguard的支持

numaga94, 2023-11-12:
### 提议
请针对mt798x的后续更新中，添加对wireguard的支持

### 为什么要
wg的性能，对有内网穿透需求的小型办公室网络实在是太重要了。

### 实例
租用的办公楼里，想要实现稳定的内网穿透，实现多个异地办公室可以同享一个nas，并拥有高效的网络传输效率。

*例如别人固件或者视频中演示的效果*

miaoermua, 2023-11-13:
关于 mt798x 对 wireguard 的支持是一直都有的 https://github.com/search?q=repo%3Amiaoermua%2FCatWrt%20wireguard&type=code，需要在软件源手动进行安装。

https://www.miaoer.xyz/posts/network/catwrt-install-application

closed 2023-11-13 by miaoermua

numaga94, 2023-11-13:
使用opkg安装wireguard，在redmi ax6000出现这样的报错：

Unknown package 'kmod-wireguard'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.134-1-29c7378c7e1adfd339581af2c1ddc93a) for kmod-crypto-lib-chacha20
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-chacha20 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.134-1-29c7378c7e1adfd339581af2c1ddc93a) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.134-1-29c7378c7e1adfd339581af2c1ddc93a) for kmod-crypto-lib-poly1305
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-poly1305 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.134-1-29c7378c7e1adfd339581af2c1ddc93a) for kmod-crypto-lib-chacha20poly1305
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-chacha20poly1305 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.134-1-29c7378c7e1adfd339581af2c1ddc93a) for kmod-crypto-kpp
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-kpp found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.134-1-29c7378c7e1adfd339581af2c1ddc93a) for kmod-crypto-lib-curve25519
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-curve25519 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.134-1-29c7378c7e1adfd339581af2c1ddc93a) for kmod-udptunnel4
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-udptunnel4 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.134-1-29c7378c7e1adfd339581af2c1ddc93a) for kmod-udptunnel6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-udptunnel6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.134-1-29c7378c7e1adfd339581af2c1ddc93a) for kmod-wireguard
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-wireguard found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-wireguard.

不知道有没有ax 6000适配的源呢？

miaoermua, 2023-11-14:
> 使用opkg安装wireguard，在redmi ax6000出现这样的报错：
>
> Unknown package 'kmod-wireguard'.
> Collected errors:
>  * pkg_hash_check_unresolved: cannot find dependency kernel (= [...]

这个是因为你没更新到最新稳定版 v23.8 所以内核版本对不上，无法安装。

请参考:
https://www.miaoer.xyz/posts/network/catwrt-applist