miaoermua/CatWrt #34: glinet mt3000 没有无线网络

lovefrr, 2024-02-20:
glinet mt3000 刷了 CatWrt.v23.8.mt798x-filogic-glinet_gl-mt3000-squashfs-sysupgrade.bin 没有无线网络，请问是不支持吗？

heartbreaker0, 2024-02-21:
GL-MT3000相同版本遇到相同问题

heartbreaker0, 2024-02-21:
在内核日志看到了这几条
`[   14.227162] mt798x-wmac 18000000.wifi: Direct firmware load for mediatek/mt7981_rom_patch.bin failed with error -2
[   14.237561] mt798x-wmac 18000000.wifi: Falling back to sysfs fallback for: mediatek/mt7981_rom_patch.bin
[   14.272809] mt798x-wmac: probe of 18000000.wifi failed with error -12`

heartbreaker0, 2024-02-21:
在内核日志看到了这几条：
[   14.227162] mt798x-wmac 18000000.wifi: Direct firmware load for mediatek/mt7981_rom_patch.bin failed with error -2
[   14.237561] mt798x-wmac 18000000.wifi: Falling back to sysfs fallback for: mediatek/mt7981_rom_patch.bin
[   14.272809] mt798x-wmac: probe of 18000000.wifi failed with error -12

miaoermua, 2024-02-21:
用 v24.3 再试试

miaoermua, 2024-02-21:
> > 用 v24.3 再试试
> 
> 24.3发布了吗，bin在哪下载呢？

群里发了，tg 需要艾特我一下备注型号。github 需要合并上去才发行，不然就乱套了。

closed 2024-02-21 by lovefrr