miaoermua/CatWrt #48: 建议更新一下tailscale的版本.

dongshimou, 2025-03-17:
### 提议
headscale 更新的版本,不支持 1.62以前的tailscale.
xiaomi AX6000 目前还是很老的版本
```
tailscale --version
1.42.0-ERR-BuildInfo
  go version: go1.20.2
```

### 为什么要
现在的 mt798x 的tailscale 已经无法连接到最新的headscale 服务端

### 实例

headscale对应的错误日志
```
2025/03/17 03:12:27 http: response.Write on hijacked connection from fmt.Fprintln (print.go:305)
2025-03-17T03:12:27Z ERR http internal server error error="noise upgrade failed: unsupported client version: 62" code=500
```

*例如别人固件或者视频中演示的效果*

miaoermua, 2025-03-17:
tailscale 只需要我们内核 kmod 支持好你可以下载 ipk 自行更新

例如

https://mirrors.pku.edu.cn/immortalwrt/releases/24.10.0/packages/aarch64_cortex-a53/packages/tailscale_1.80.2-r1_aarch64_cortex-a53.ipk

这里用 cattools > 更多 > 导入ipk > 将上诉 url 粘贴 > 选择尝试使用 opkg update 就安装成功了

附 cattools 使用教程：https://www.miaoer.net/posts/blog/cattools

https://www.miaoer.net/posts/blog/cattools-step

miaoermua, 2025-03-24:
这里没有问题继续提出就暂时关闭此 issues 了

closed 2025-03-24 by miaoermua