miaoermua/CatWrt #52: 关于AX6000升级中问题和请求

CrazyJasonwell, 2025-05-04:
### 提供 BUG 问题日志

```markdown
我使用大佬的固件重刷了AX6000,版本号23.2.14，使用存在一些问题：
1 插件不能升级，软件源配置有问题

Downloading https://mirrors.cloud.tencent.com/lede/snapshots/targets/mediatek/filogic/packages/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/targets/mediatek/filogic/packages/Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/base/Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/releases/18.06.9/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://mirrors.cloud.tencent.com/lede/releases/18.06.9/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz

Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
*** Failed to download the package list from https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz

tencent镜像站点没有这个目录 

2 由于不能升级插件，很多软件不能使用，比如nezha版本低了，不能连接远程的dashboard,openclash版本了一直报redir-host模式错误等等 

3 这个23.2版本的固件是否存在升级版本，告知是否能升级到新的固件CatWrt-v23.8 mt798x，不敢随便更新，一旦更新失败家里就会没有网络。

请大佬抽空回复我一下，谢谢！
```

### 你的设备

Xiaomi Redmi Router AX6000 (CpuMark : 27241.208519 Scores)

### CatWrt_Version

v23.2 (AMD64 & mt798x & Rock64)

### CatWrt_Arch

mt798x

### 能否复现 BUG

能复现

### 最后确认

- [x] 我确认我的 BUG 来自 CatWrt，而不是其他 OpenWrt 固件
- [x] 我将遵守 CatWrt 开源社区规范，不将设备用为商业目的
- [x] 我已确认我的内容反馈内容的正确

miaoermua, 2025-05-04:
需要把软件源切换到 CatWrt 源，默认是上游通用的源，不太能用。

你这个版本是 v23.2 可以参考一下 https://github.com/miaoermua/cattools/tree/main/repo/history/v23.2/mt7986a

Cattools 不一定能用，另外相关教程是

https://www.miaoer.net/posts/network/catwrt-install-application

https://www.miaoer.net/posts/network/catwrt-applist

使用对应的软件源，如果你要使用到国内服务器也不妨可以稍微支持一下该项目的健康发展 https://www.miaoer.net/sponsor

miaoermua, 2025-05-05:
可以直接升级到 v23.8

CrazyJasonwell, 2025-05-05:
谢谢你的回复。
按照你的指点，我做了几个事情：
1 修改软件源后完成了插件的更新，openclash升级到了0.46.079(官网最新),nezha-agent只能升级luci-app-nezha-agent到2.0,nezha-agent还是1.1版，frpc还是0.47不能升级到最新的0.62

2 Cattools暂时还没有用，回头抽空安装试用一下

3 升级固件23.8,遇到一些问题，请大佬关注一下：我首先备份了配置，然后上传23.8固件升级（保留配置），然后自动重启后能正常登录，只是不能上网，后来又修改软件源，等等，又重启了一下还是不能上网了，而且发现服务菜单里面只有几个插件了，openclash、nezha、frp、smartdns、passwall等等都没有显示，估计是不能连接软件源，导致更新插件失败，不能上网的原因怀疑可能是使用了原来的配置（openclash影响），插件又不能更新，成了一个死循环，只好又刷回23.2版本。由于保留了配置，很快恢复之前的状态了。

4 咨询一下：我下载的ipk文件能不能手动安装：opkg install XXX.ipk，主要是luci有对应的UI包和配置数据，不知道放到哪里？原来的版本是不是有一个上传软件包，然后点击安装的界面，这样的话就不用担心配置数据混乱了。另外，由于硬路由空间小，升级过程的中间文件哪些是可以删除的，也请指导一下。还有就是openwrt下的openclash出去网速感觉比win/mac客户端部署的clash要慢一些，是因w为路由器硬件配置太低的缘故吗？

5 23.8版固件升级问题，请大佬再看看，到底我升级过程中哪里不对？

再次表示感谢！

miaoermua, 2025-05-05:
你好，升级过程会放弃当前版本预装的包改为新版预装的包，所以是需要通过软件源进行安装的。

v23.8 我们做了精简想测试一下多设备兼容，nezha 我这里因为都已经部署好了是 v0.x 的版本，所以我这里不提供 v1.x 的情况，你可以自行安装 ipk 尝试。

Cattools 可以切换软件源，可以安装 ipk，具体你可以参考一下哦，不会占用很多空间本身就是一个一千多行的脚本。

miaoermua, 2025-05-16:
如果没有问题我这里先标记完成，如果你有任何问题也可以随时打开

closed 2025-05-16 by miaoermua