miaoermua/CatWrt #53: 包 `nfs-utils` 支持

lovemilk2333, 2025-05-30:
### 提议
提供 `kmod-dm`, 使得 `nfs-utils` 可以被安装

### 为什么要
```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-dm for libdevmapper-selinux
 * pkg_hash_check_unresolved: cannot find dependency kmod-dm for libdevmapper
 * pkg_hash_fetch_best_installation_candidate: Packages for libdevmapper found, but incompatible with the architectures configured
```

### 实例

[image]

miaoermua, 2025-07-16:
https://github.com/miaoermua/catwrt_base/blob/main/build/config/amd64.config#L6232

miaoermua, 2025-07-17:
https://github.com/miaoermua/CatWrt/releases/tag/v25.8

https://github.com/miaoermua/CatWrt/pull/55

closed 2025-07-17 by miaoermua