miaoermua/CatWrt #54: dhcp/dns设置界面进行静态地址分配后，状态-概览-dhcp分配页面和其他显示设备设备名称的地方会变为Infinte或者?

iko233, 2025-07-06:
### 提供 BUG 问题日志

```markdown
`日志没有看到什么比较有用的信息`
```

### 关于这个问题 BUG 描述

dhcp/dns设置界面进行静态地址分配后，状态-概览-dhcp分配页面和其他显示设备设备名称的地方会变为'Infinte'或者'?',似乎最后一个增加静态地址的会显示为'Infinte'，其他的会显示为'?'.[image]

[image],

### 你的设备

WTM W-P95/W-P95 - Intel(R) Celeron(R) N5095 @ 2.00GHz : 4C4T 

### CatWrt_Version

v24.9 (AMD64 & mt7621)

### CatWrt_Arch

AMD64 (x86_64)

### 能否复现 BUG

能复现

### 最后确认

- [x] 我确认我的 BUG 来自 CatWrt，而不是其他 OpenWrt 固件
- [x] 我将遵守 CatWrt 开源社区规范，不将设备用为商业目的
- [x] 我已确认我的内容反馈内容的正确

miaoermua, 2025-07-07:
不建议设置租期为 12h 的其他值，超过会有问题，另外设置设备名字通过 MAC 地址，地址为辅(用于确定设备)，使用网络 - 主机名例如：

[image]

miaoermua, 2025-07-07:
抱歉，昨天没看到，没及时回复。

如果问题已经通过我的解决方法确定没问题了，请自行关闭 issues。 🤗

closed 2025-07-07 by iko233

iko233, 2025-07-08:
问题原因确定了,我有些疏忽了,给你增添麻烦了，租期使用了错误的拼写Infinte导致的问题。
问题来自上游项目(luci)页面没有对值进行校验. 但是上游项目luci已经转向js去渲染页面了，
这边的项目停留在老版本lua渲染页面的方式,luci那边应该是没有维护了,是否考虑拆包ipk修改这个问题？

修改范围luci-mod-admin-full_git-24.182.36151-66d29de-1_x86_64.ipk-dhcp.lua 和  (i18n文件)

miaoermua, 2025-07-08:
> [@miaoermua](https://github.com/miaoermua) 问题原因确定了,我有些疏忽了,给你增添麻烦了，租期使用了错误的拼写Infinte导致的问题。 问题来自上游项目(luci)页面没有对值进行校验. 但是上游项目luci已经转向js去渲染页面了， 这边的项目停留在老版本lua渲染页面的方式,luci那边应该是没有维护了,是否考虑拆包ipk修改这个问题？
> 
> 修改范围luci-mod-admin-full_git-24.182.36151-66d29de-1_x86_64.ipk-dhcp.lua 和 (i18n文件)

确实有一段时间了，你可以看看，他这里也有分支几个。
https://github.com/coolsnowwolf/luci/tree/master/modules/luci-mod-admin-full